daqifi/daqifi-desktop #29: Channels Remain After Abrupt Device Disconnect

cptkoolbeenz, 2023-06-08:
If a device disconnects unexpectedly, the previously enabled channels remain after reconnecting a device.

To reproduce:
Connect to a device via USB, enable some channels.  Disconnect device from USB.  Reconnect to device and check channels tab.  The previously configured channels are still listed and cannot be deleted.

codedecoded01723, 2024-12-06:
@cptkoolbeenz , Serial device unexpectedly disconnect is working fine and we have implemented the popup also and  when it does, it shows the the message by saying the device is unexpectedly disconnected , and clears out the channels of it as well, NOW
we have implemented this in devices connected through WIFI as well but there is a catch , when we power off the device from the power button on device , the IP of device still pings from the backend or CMD and when we completely cut the supply from power switch it stops pinging ,  So it get hard to detect the device is active or not if it is power off from the device power button.

cptkoolbeenz, 2024-12-06:
Ah interesting.  It likely stems from the new WiFi implementation staying active after the device is told to power down while connected via USB.  I'm guessing if you disconnect from USB then test WiFi ping after power-off, it should be silent.  I will make a ticket in the firmware to shut down the WiFi module during power-down.

cptkoolbeenz, 2024-12-21:
I think the latest beta release has this taken care of.  @codedecoded01723 can you verify in the latest beta firmware?