daqifi/daqifi-desktop #31: Add Differential Channel Configuration in Channel Settings

cptkoolbeenz, 2023-06-12:
Nq1 channels are ground-referenced single-ended by default.  Channels 8+9, 10+11, 12+13, 14+15, can be optionally used as differential channels.  The advertised differential capability and subsequent setting is represented by analog_in_port_av_rse and analog_in_port_rse protobuf tags respectively.

The app should query the differential capably and provide the user an option to configure a channel pair as RSE or differential.

MarshBod, 2023-10-11:
Hi just a few questions on this topic:
1. You mention channels 8+9, 10+11, 12+13, 14+15 for differential outputs, however this does not correspond to the data sheet for the Nyquist 1 so which channels are supposed to be differential?
2.  When trying the recommended ports for differential readings, how to i apply this in the software, do i add a singular channel out of the pair and set it to differential or do i need to apply both channels which are connected?
3. Does this Nyquist 1 have a built-in amplification module for reading scaling the mV signals of load cells, if not what is the recommended reading voltages for the analogue ports

Also, there was mention of a firmware update pending I am wondering as to where i can find this update.

cptkoolbeenz, 2023-10-11:
1. Yes, the datasheet needs to be updated.  The hardware supports 8+9, 10+11, 12+13, 14+15 for optional differential inputs.
2. The Windows application needs to be updated to properly configure differential channels.  The SCPI commands for differential configuration are working but the output is currently not converted from 2's complement, so the resulting number is a direct integer based on the 2's complement value coming from the ADC.  This will be addressed in firmware/software.
3. It does not.  Load cell signals should be passed through something like the [EMBSGB200](https://tacunasystems.com/products/amplifiers-conditioners/tacuna-systems-load-cell-amplifier/) or other bridge amplifier and sent to the DAQ.

tylerkron, 2025-04-11:
Current hardware doesn't support differential measurements.  Other versions will.