daqifi/daqifi-desktop #95: WiFi Firmware update progress bar misleading

cptkoolbeenz, 2025-02-01:
Updating firmware seems to follow the progress bar correctly.  However, the subsequent WiFi update progress bar appears to complete very quickly.  Much later, the firmware update complete prompt fires.

tylerkron, 2025-02-02:
@cptkoolbeenz - I don't think a second progress bar should exist. Just update the main one. Thoughts?

cptkoolbeenz, 2025-02-02:
WiFi firmware should seldom need updating...  So, we might want to decouple the firmware from WiFi firmware updating.  Added sub issue related to that.

tylerkron, 2025-02-03:
@cptkoolbeenz - My thought is the end-user shouldn't even care/know that the firmware and wifi firmware are two different things. It'd be odd if my Google Nest was notifying me of component updates. To them the "Firmware" should be all encompassing software that runs on the board.

cptkoolbeenz, 2025-02-03:
Yeah I think that would be good for UI as long as we have a log and/or maybe display what step of the process it is on. That way we can easily debug when there is an issue.

cptkoolbeenz, 2025-02-07:
Maybe we can read the output stream from winc_flash_tool.cmd and count the number of "e","w","r","and "v"s to update the progress bar?

```
begin erase operation

 0x000000:[eeeeeeee] 0x008000:[eeeeeeee] 0x010000:[eeeeeeee] 0x018000:[eeeeeeee]
 0x020000:[eeeeeeee] 0x028000:[eeeeeeee] 0x030000:[eeeeeeee] 0x038000:[eeeeeeee]
 0x040000:[eeeeeeee] 0x048000:[eeeeeeee] 0x050000:[eeeeeeee] 0x058000:[eeeeeeee]
 0x060000:[eeeeeeee] 0x068000:[eeeeeeee] 0x070000:[eeeeeeee] 0x078000:[eeeeeeee]

begin write operation

 0x000000:[wwwwwwww] 0x008000:[wwwwwwww] 0x010000:[wwwwwwww] 0x018000:[wwwwwwww]
 0x020000:[wwwwwwww] 0x028000:[wwwwwwww] 0x030000:[wwwwwwww] 0x038000:[wwwwwwww]
 0x040000:[wwwwwwww] 0x048000:[wwwwwwww] 0x050000:[wwwwwwww] 0x058000:[wwwwwwww]
 0x060000:[wwwwwwww] 0x068000:[wwwwwwww] 0x070000:[wwwwwwww] 0x078000:[wwwwwwww]

begin read operation

 0x000000:[rrrrrrrr] 0x008000:[rrrrrrrr] 0x010000:[rrrrrrrr] 0x018000:[rrrrrrrr]
 0x020000:[rrrrrrrr] 0x028000:[rrrrrrrr] 0x030000:[rrrrrrrr] 0x038000:[rrrrrrrr]
 0x040000:[rrrrrrrr] 0x048000:[rrrrrrrr] 0x050000:[rrrrrrrr] 0x058000:[rrrrrrrr]
 0x060000:[rrrrrrrr] 0x068000:[rrrrrrrr] 0x070000:[rrrrrrrr] 0x078000:[rrrrrrrr]

verify range 0x000000 to 0x080000
begin verify operation

 0x000000:[vvvvvvvv] 0x008000:[vvvvvvvv] 0x010000:[vvvvvvvv] 0x018000:[vvvvvvvv]
 0x020000:[vvvvvvvv] 0x028000:[vvvvvvvv] 0x030000:[vvvvvvvv] 0x038000:[vvvvvvvv]
 0x040000:[vvvvvvvv] 0x048000:[vvvvvvvv] 0x050000:[vvvvvvvv] 0x058000:[vvvvvvvv]
 0x060000:[vvvvvvvv] 0x068000:[vvvvvvvv] 0x070000:[vvvvvvvv] 0x078000:[vvvvvvvv]
```

tylerkron, 2025-04-04:
Fixed in the latest release. still 2 bars which I don't like but the progress updates based on `begin write operation`, `begin read operation`, and `begin verify operation`

closed 2025-04-04 by tylerkron