daqifi/daqifi-desktop #102: User Account Control trigger

cptkoolbeenz, 2025-02-02:
Each time DAQiFi is opened it triggers a Windows UAC prompt.  Is this possible to get permission once and forever to prevent future prompts?

cptkoolbeenz, 2025-02-06:
NOTE: It seems like the x86 version doesn't trigger the UAC prompt upon each run of the application.

cptkoolbeenz, 2025-04-11:
Looks like Sectigo is 3 Year @ $219.45 per year.

tylerkron, 2025-04-11:
There is another way to do this too

## Microsoft Trusted Signing vs. Traditional Certificates for Your Scenario

### Integration with GitHub CI/CD
Microsoft Trusted Signing offers direct integration with GitHub Actions through the `azure/trusted-signing-action@v0` GitHub Action[4]. This allows you to automatically sign your application during your existing CI/CD workflow with code like:

```yaml
- name: Sign files with Trusted Signing
  uses: azure/trusted-signing-action@v0
  with:
    azure-tenant-id: ${{ secrets.AZURE_TENANT_ID }}
    azure-client-id: ${{ secrets.AZURE_CLIENT_ID }}
    azure-client-secret: ${{ secrets.AZURE_CLIENT_SECRET }}
    endpoint: https://eus.codesigning.azure.net/
    trusted-signing-account-name: your-account-name
    certificate-profile-name: your-certificate-profile
    files-folder: ${{ github.workspace }}\App\bin\Release\net8.0-windows
    files-folder-filter: exe,dll
```

### Cost Comparison
- **Trusted Signing**: $9.99/month (Basic) for up to 5,000 signatures or $99.99/month (Premium) for up to 100,000 signatures
- **OV Certificates**: $129-$380/year 
- **EV Certificates**: $269-$699/year

For infrequent releases, Trusted Signing could be more cost-effective than traditional certificates.

### Setup Requirements
While you don't currently use Azure, to use Trusted Signing you will need:
1. An Azure subscription
2. A Microsoft Entra tenant ID
3. Registration of the Microsoft.CodeSigning resource provider
4. Identity validation through the Azure portal (organizations must have 3+ years of verifiable business history)[3][7]

### Key Advantages for Your Scenario
1. **Automated Certificate Management**: No need to worry about certificate renewal or secure storage - certificates are short-lived (72 hours) and automatically renewed[12]
2. **SmartScreen Reputation**: Trusted Signing provides reputation with Microsoft SmartScreen, similar to EV certificates
3. **Security**: Private keys are stored in FIPS 140-2 Level 3 certified HSMs[9]
4. **Seamless GitHub Integration**: Direct integration with your existing CI/CD pipeline

### Potential Drawbacks
1. **Azure Dependency**: You'll need to set up and maintain Azure resources
2. **Monthly Subscription**: Unlike one-time certificate purchases, Trusted Signing requires ongoing payment
3. **Identity Validation Process**: You'll need to complete the validation process in the Azure portal

## Recommendation

Trusted Signing makes more sense than traditional OV/EV certificates for your scenario if:
1. You release updates frequently (making the monthly subscription model cost-effective)
2. You want to avoid the hassle of certificate management and security
3. You prefer integration with your GitHub workflows over manual signing
4. You want immediate SmartScreen reputation without paying for an expensive EV certificate

If you release updates infrequently and want to avoid setting up Azure resources, a traditional OV certificate might be simpler, though it won't provide immediate SmartScreen reputation like Trusted Signing or an EV certificate would.

Citations:
[1] https://pplx-res.cloudinary.com/image/upload/v1744397588/user_uploads/rTvFHdqbqtkwqfp/image.jpg
[2] https://learn.microsoft.com/en-us/azure/trusted-signing/how-to-signing-integrations
[3] https://learn.microsoft.com/en-us/azure/trusted-signing/quickstart
[4] https://github.com/marketplace/actions/trusted-signing
[5] https://support.ksoftware.net/support/solutions/articles/232500-what-are-the-differences-between-ov-and-ev-level-code-signing-certificates-
[6] https://signmycode.com/blog/trusted-signing-the-microsoft-way-to-safeguard-your-applications
[7] http://www.lindersoft.com/forums/showthread.php?48342-SetupBuilder-2025-and-Microsoft-Trusted-Signing
[8] https://www.ssl2buy.com/wiki/regular-code-signing-vs-ev-code-signing
[9] https://learn.microsoft.com/en-us/azure/trusted-signing/overview
[10] https://www.certauri.com/comparing-code-signing-certificate-prices-your-guide-2/
[11] https://www.cheapsslshop.com/blog/ev-code-signing-vs-ov-code-signing-certificate
[12] https://learn.microsoft.com/en-us/azure/trusted-signing/concept-trusted-signing-cert-management
[13] https://learn.microsoft.com/en-us/answers/questions/1460330/signing-wpf-application-for-distribution
[... 5 lines cut ...]
[19] https://learn.microsoft.com/en-us/answers/questions/2143781/quota-for-azure-trusted-signing
[20] https://learn.microsoft.com/en-us/answers/questions/1858768/i-azure-trusted-signing-fast-and-reliable
[21] https://learn.microsoft.com/en-us/dotnet/desktop/wpf/migration/?view=netdesktop-9.0
[22] https://www.advancedinstaller.com/continuous-signing-ci-cd-best-practices.html
[23] https://clarionhub.com/t/microsoft-trusted-code-signing-steps-to-set-it-up/7861
[24] https://www.reddit.com/r/dotnet/comments/1115foi/how_difficult_is_to_get_a_trusted_key_for_signing/
[25] https://www.hendrik-erz.de/post/code-signing-with-azure-trusted-signing-on-github-actions
[26] https://learn.microsoft.com/en-us/answers/questions/2045602/azure-trusted-signing-how-can-i-digitally-sign-a-v
[27] https://www.sectigo.com/ssl-certificates-tls/code-signing
[28] https://www.reddit.com/r/electronjs/comments/17sizjf/a_guide_to_code_signing_certificates_for_the/
[29] https://azure.microsoft.com/en-us/pricing/details/trusted-signing/
[30] https://www.ssl.com/faqs/which-code-signing-certificate-do-i-need-ev-ov/
[... cut ...]
[35] https://forum.juce.com/t/signing-windows-installers-with-ov-certificate-how-long-to-get-smartscreen-reputation/55245
[36] https://www.entrust.com/resources/learn/compare-ssl-certificates-ev-vs-ov
[37] https://www.reddit.com/r/dotnet/comments/13mnvh6/ev_code_signing/
[38] https://techcommunity.microsoft.com/blog/microsoft-security-blog/simplifying-signing-integration-for-trusted-signing/4293292
[39] https://www.reddit.com/r/AZURE/comments/1buni1p/azure_trusted_signing/
[40] https://github.com/Azure/trusted-signing-action

---
Answer from Perplexity: pplx.ai/share