daqifi/daqifi-desktop #126: Import logging session from device to application

tylerkron, 2025-03-09:
In #65 we added the ability for the user to command the board to log to the device instead of steam to the application. It also added the ability to list the logging sessions on the device.

Now we want to be able to import the device logging sessions to the application so the user can view them and download them.

cptkoolbeenz, 2025-11-04:
Note, SD files can be any type of streaming format the Nyquist allows (protobuf, json, csv).  So, the app would need to determine the filetype.

I only expect protobuf files would be imported to the application where json and csv files would have the option to download to the PC as a plain file.

tylerkron, 2026-03-16:
Resolved by #375 — SD card log files can now be imported from a connected USB device or from a local .bin file, with proper voltage scaling and timestamp reconstruction via daqifi-core v0.19.2.