daqifi/daqifi-desktop #168: Windows App sends SYSInfoPB? three times during connection

cptkoolbeenz, 2025-07-20:
## Description
During device connection, the Windows desktop application sends the  SCPI command exactly three times in succession, even when receiving valid responses.

## Steps to Reproduce
1. Connect to a DAQiFi device via USB using the Windows desktop app
2. Monitor serial communication with Wireshark or similar tool
3. Observe the connection sequence

## Expected Behavior
The application should send  once and proceed with the connection after receiving a valid response.

## Actual Behavior
The application sends  three times, receiving identical valid responses each time before proceeding.

## Technical Details
- All three responses contain valid protobuf data with correct field values:
  - Field 17 (analog_in_port_num) = 16
  - Field 35 (digital_port_num) = 16
  - Field 38 (analog_out_port_num) = 0
- The application eventually proceeds normally after the third response
- This behavior adds unnecessary delay to the connection process

## Impact
- Low severity - connection eventually succeeds
- Adds ~300-500ms unnecessary delay to connection time
- Could cause buffer issues with very large protobuf messages in the future

## Additional Context
This was discovered while debugging connection issues. The firmware sends valid responses all three times, suggesting the issue is in the desktop application's connection logic.

cptkoolbeenz, 2025-07-23:
## Update: Root Cause Identified - False Alarm

After investigation in the firmware, this behavior is **by design** and not a bug. The Windows app is correctly waiting for the device to fully power up.

### What is happening:
1. When the device first connects, it starts in **STANDBY** state (power state = 1)
2. The device needs time to transition to **RUN** state (power state = 2)
3. This transition typically happens around the 3rd retry

### Test Results:
```
Attempt 1: State: STANDBY (1)  < /dev/null |  Battery: 0.00V
Attempt 2: State: STANDBY (1) | Battery: 0.00V  
Attempt 3: State: RUN (2) | Battery: 4.07V
```

The Windows app is correctly polling until:
- Power state reaches 2 (POWERED_UP/RUN)
- Battery voltage reads > 0V (indicating full power-up)

### Recommendation:
Instead of the current rapid-fire approach, consider adding a **300-500ms delay** between SYSInfoPB attempts. This would:
- Reduce unnecessary USB traffic
- Give the device time to power up between checks
- Still detect the state change promptly

The 3-retry behavior is actually correct and ensures the device is fully powered before proceeding.

🤖 Generated with [Claude Code](https://claude.ai/code)

tylerkron, 2026-03-20:
Closing — the multiple `SYSInfoPB?` sends on serial connections are now by design.

After the Core library migration, the serial connection path intentionally retries `SYSInfoPB?` every 1 second (up to 8s) in `WaitForInitialStatusMessage()` until the device responds. If the device takes ~2s to reply, you'll see exactly 3 sends — matching this report.

WiFi connections only send it once. The serial retry logic is a reliability mechanism for the less reliable serial transport.

closed 2026-03-20 by tylerkron