daqifi/daqifi-desktop #203: feat: Add multi-plot support with configurable channel assignment

cptkoolbeenz, 2025-08-12:
## Summary
Add the ability to display channels on different plot panes, allowing users to configure which channels appear on which plot. This would enable better visualization of data with different scales, units, or measurement types.

## Current Behavior
All channels are displayed on a single plot view, which can make it difficult to:
- Compare channels with vastly different scales
- Visualize different measurement types (e.g., voltage vs temperature)
- Focus on specific channel groups
- Analyze correlations between specific channel subsets

## Proposed Feature
Implement a multi-plot system where users can:
1. Create multiple plot panes (e.g., 2-4 plots stacked vertically or in a grid)
2. Assign channels to specific plots via configuration
3. Each plot maintains its own:
   - Y-axis scale and range
   - Grid settings
   - Zoom level (optionally synchronized X-axis)
   - Color assignments for its channels

## User Interface Concept
### Configuration Options:
- **Plot Layout**: Choose between single, dual, triple, or quad plot views
- **Channel Assignment**: Drag-and-drop or checkbox interface to assign channels to plots
- **Plot Settings**: Individual settings per plot (scale, grid, labels)
- **Synchronization**: Option to link/unlink X-axis (time) across plots

### Example Use Cases:
1. **Mixed Signals**: Plot analog inputs on Plot 1, digital I/O on Plot 2
2. **Different Scales**: High voltage channels on Plot 1, low voltage sensors on Plot 2
3. **Grouped Analysis**: Temperature sensors on Plot 1, pressure sensors on Plot 2
4. **Comparison**: Reference signal on Plot 1, test signals on Plot 2

## Technical Considerations
- Maintain performance with multiple plots rendering simultaneously
- Share time axis to ensure synchronized data viewing
- Support existing features (zoom, pan, export) for each plot
- Save/load plot configurations with profiles
- Handle dynamic channel addition/removal

## Mockup/Visualization
```
┌─────────────────────────────────────────┐
│ Plot 1: Voltage Channels (AI0-AI3)     │
│ [Graph showing voltage waveforms]       │
├─────────────────────────────────────────┤
│ Plot 2: Current Channels (AI4-AI7)     │
│ [Graph showing current measurements]    │
├─────────────────────────────────────────┤
│ Plot 3: Digital I/O (DI0-DI3)          │
│ [Graph showing digital signals]         │
└─────────────────────────────────────────┘
```

## Benefits
- Improved data visualization for complex measurements
- Better channel organization and grouping
- Easier analysis of channels with different characteristics
- Enhanced user experience for multi-channel data acquisition

## Acceptance Criteria
- [ ] Users can create 1-4 plot panes
- [ ] Channels can be assigned to specific plots
- [ ] Each plot has independent Y-axis scaling
- [ ] X-axis (time) can be synchronized across plots
- [ ] Configuration can be saved and loaded
- [ ] Performance remains acceptable with multiple plots
- [ ] Existing single-plot mode remains available

## Related Issues
- #147 - Overview Minimap for Large Dataset Navigation
- #85 - Allow user to show/hide channels on live plot

cc: @weehau1984 @tylerkron - This feature would significantly enhance the data visualization capabilities. Would love to get your thoughts on the implementation approach and UI/UX considerations.

🤖 Generated with [Claude Code](https://claude.ai/code)

tylerkron, 2025-08-12:
Man you getting fancy up in here. LFG 🚀

cptkoolbeenz, 2025-08-12:
As long as we have AI slaves...